thelastoutpostworkshop/gpio_viewer #151: Runtime Error

RomanPittroff, 2025-01-07:
Hello 

Tried to implement your gpio viewer with an ESP 32 S4 Dev board.

- v3.1.0 [ESP32 Arduino Core](https://github.com/espressif/arduino-esp32)
- v3.3.1 [Async TCP](https://github.com/mathieucarbou/AsyncTCP)
- v3.4.5 [ESP Async WebServer](https://github.com/mathieucarbou/ESPAsyncWebServer)

And get follow error in the serial monitor.
`assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)`

then follow a reboot.

Downgrade to  
- v3.0.7 [ESP32 Arduino Core](https://github.com/espressif/arduino-esp32)

And this was working 

> GPIOViewer >> Release 1.5.8
> GPIOViewer >> ESP32 Core Version 3.0.7
> GPIOViewer >> Chip Model:ESP32-S3, revision:2
> GPIOViewer >> No PSRAM
> GPIOViewer >> Web Application URL is: http://192.168.1.114:8080

But then the URL can be open also the ping IP shows a Request timed out

thelastoutpostworkshop, 2025-01-07:
The error comes from the ESPAsyncWEbServer library [issue](https://github.com/espressif/arduino-esp32/issues/10526) with v3.1.0 of Espressif Core.
v3.4.5 of this library should have fixed this problem.

thelastoutpostworkshop, 2025-01-18:
Stale issue, closing

closed 2025-01-18 by thelastoutpostworkshop