thelastoutpostworkshop/gpio_viewer #152: ESP32-Wroom-32 non fatal compile time error - informational

andyp60, 2025-01-12:
I compiled the GPIOViewer supplied example with the following output, with fatal results. I deleted and reloaded all relevant libraries, noted below, showing recommended versions and my final versions. I still get the errors but it loads successfully and I can access the page from a browser and monitor pin status.

Arduino IDE Version: 2.3.4
Date: 2024-12-03T10:03:03.490Z
CLI Version: 1.1.1

// Espressif Arduino Core v3.1.0 - 3.1.1
// ESP Async WebServer 3.4.5 - 3.6.0
// AsyncTCP            3.3.1 -3.3.2

In file included from C:\Users\andyp\AppData\Local\Temp\.arduinoIDE-unsaved2025012-22600-6tjt99.fioq9\gpioviewer\gpioviewer.ino:14:
c:\Users\andyp\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h: In lambda function:
c:\Users\andyp\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:184:41: warning: 'void AsyncWebServerRequest::send_P(int, const String&, const char*, AwsTemplateProcessor)' is deprecated: Replaced by send(int code, const String& contentType, const char* content = asyncsrv::empty, AwsTemplateProcessor callback = nullptr) [-Wdeprecated-declarations]
  184 |                        { request->send_P(200, "text/html", generateIndexHTML().c_str()); });
      |                          ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\andyp\Documents\Arduino\libraries\GPIOViewer\src/gpio_viewer.h:11:
c:\Users\andyp\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:350:10: note: declared here
  350 |     void send_P(int code, const String& contentType, PGM_P content, AwsTemplateProcessor callback = nullptr) {
      |          ^~~~~~

thelastoutpostworkshop, 2025-01-12:
Thanks for letting me know, I will make the required changes for the next revision

thelastoutpostworkshop, 2025-01-14:
This as been solved in v1.6.0

closed 2025-01-14 by thelastoutpostworkshop