thelastoutpostworkshop/gpio_viewer #153: App not starting when GPIO viewer added.

atlantikus, 2025-01-12:
Hi, I'm not professional programmar, only copy coder educated from youtube. I have CYD board ESP32-2432S028 and as part of learning I tried your other project animated_gif_sdcard_spiffs.ino which worked fine. Then I tried to add GPIO Viewer as instructed in video. GPIO viewer starts, I'm able to see I/Os on website. But my code then does not continue to animated GIF.

When gpio_viewer.begin(); is removed from setup it works fine. Maybe it is beginner issue. Maybe config issue.

`#include <SD.h>
#include <SPI.h>
#include <FS.h>
#include <SPIFFS.h>
#include <TFT_eSPI.h>      // Install this library with the Arduino Library Manager
                           // Don't forget to configure the driver for the display!

#include <AnimatedGIF.h>   // Install this library with the Arduino Library Manager
#include <gpio_viewer.h>

#define SD_CS_PIN 5 // Chip Select Pin (CS) for SD card Reader

GPIOViewer gpio_viewer;

AnimatedGIF gif;
File gifFile;              // Global File object for the GIF file
TFT_eSPI tft = TFT_eSPI(); 

const char *filename = "/death_star.gif";   // Change to load other gif files in images/GIF
void setup()
{
  Serial.begin(115200);

  gpio_viewer.connectToWifi("aaaa","bbbb");

  tft.begin();
  tft.setRotation(1); // Adjust Rotation of your screen (0-3)
  tft.fillScreen(TFT_BLACK);

  // Initialize SD card
  Serial.println("SD card initialization...");
  if (!SD.begin(SD_CS_PIN))
  {
    Serial.println("SD card initialization failed!");
    return;
  }

  // Initialize SPIFFS
  Serial.println("Initialize SPIFFS...");
  if (!SPIFFS.begin(true))
  {
    Serial.println("SPIFFS initialization failed!");
  }

  // Reformmating the SPIFFS to have space if a large GIF is loaded
  // You could run out of SPIFFS storage space if you load more than one image or a large GIF
  // You can also increase the SPIFFS storage space by changing the partition of the ESP32
  //
  Serial.println("Formatting SPIFFS...");
  SPIFFS.format(); // This will erase all the files, change as needed, SPIFFs is non-volatile memory
  Serial.println("SPIFFS formatted successfully.");

  // Open GIF file from SD card
  Serial.println("Openning GIF file from SD card...");
  File sdFile = SD.open(filename);
  if (!sdFile)
  {
    Serial.println("Failed to open GIF file from SD card!");
    return;
  }

  // Create a file in SPIFFS to store the GIF
  File spiffsFile = SPIFFS.open(filename, FILE_WRITE, true);
  if (!spiffsFile)
  {
    Serial.println("Failed to copy GIF in SPIFFS!");
    return;
  }

  // Read the GIF from SD card and write to SPIFFS
  Serial.println("Copy GIF in SPIFFS...");
  byte buffer[512];
  while (sdFile.available())
  {
    int bytesRead = sdFile.read(buffer, sizeof(buffer));
    spiffsFile.write(buffer, bytesRead);
  }

  spiffsFile.close();
  sdFile.close();

  // Initialize the GIF
  Serial.println("Starting animation...");
  gif.begin(BIG_ENDIAN_PIXELS);

  gpio_viewer.begin();
}

void loop()
{
  if (gif.open(filename, fileOpen, fileClose, fileRead, fileSeek, GIFDraw))
  {
    tft.startWrite(); // The TFT chip slect is locked low
    while (gif.playFrame(true, NULL))
    {
    }
    gif.close();
    tft.endWrite(); // Release TFT chip select for the SD Card Reader
  }
}

// Callbacks for file operations for the Animated GIF Lobrary
void *fileOpen(const char *filename, int32_t *pFileSize)
{
  gifFile = SPIFFS.open(filename, FILE_READ);
  *pFileSize = gifFile.size();
  if (!gifFile)
  {
    Serial.println("Failed to open GIF file from SPIFFS!");
  }
  return &gifFile;
}

void fileClose(void *pHandle)
{
  gifFile.close();
}

int32_t fileRead(GIFFILE *pFile, uint8_t *pBuf, int32_t iLen)
{
  int32_t iBytesRead;
  iBytesRead = iLen;
  if ((pFile->iSize - pFile->iPos) < iLen)
    iBytesRead = pFile->iSize - pFile->iPos;
  if (iBytesRead <= 0)
    return 0;

  gifFile.seek(pFile->iPos);
  int32_t bytesRead = gifFile.read(pBuf, iLen);
  pFile->iPos += iBytesRead;

  return bytesRead;
}

int32_t fileSeek(GIFFILE *pFile, int32_t iPosition)
{
  if (iPosition < 0)
    iPosition = 0;
  else if (iPosition >= pFile->iSize)
    iPosition = pFile->iSize - 1;
  pFile->iPos = iPosition;
  gifFile.seek(pFile->iPos);
  return iPosition;
}
`

thelastoutpostworkshop, 2025-01-14:
The gpio header file must be the first header file included:
```
#include <gpio_viewer.h>
#include <SD.h>
#include <SPI.h>
#include <FS.h>
#include <SPIFFS.h>
#include <TFT_eSPI.h> // Install this library with the Arduino Library Manager
// Don't forget to configure the driver for the display!

#include <AnimatedGIF.h> // Install this library with the Arduino Library Manager
```
Do  you have the same problem after making this change ?

atlantikus, 2025-01-14:
I tried to place gpio_viewer.h as suggested to top of code

#include <gpio_viewer.h>
#include <SD.h>
#include <SPI.h>
#include <FS.h>
#include <SPIFFS.h>
#include <TFT_eSPI.h>      // Install this library with the Arduino Library Manager
                           // Don't forget to configure the driver for the display!

#include <AnimatedGIF.h>   // Install this library with the Arduino Library Manager

It still does not start. Serial monitor shows following:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4688
load:0x40078000,len:15460
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3196
entry 0x400805a4
GPIOViewer >> Connecting to WiFi...
.GPIOViewer >> Connected to WiFi
SD card initialization...
Initialize SPIFFS...
Formatting SPIFFS...
E (2325) task_wdt: esp_task_wdt_reset(705): task not found
E (2326) task_wdt: esp_task_wdt_reset(705): task not found
E (2351) task_wdt: esp_task_wdt_reset(705): task not found
E (2352) task_wdt: esp_task_wdt_reset(705): task not found
E (2352) task_wdt: esp_task_wdt_reset(705): task not found

~500 times E (xxx) task_wdt: esp_task_wdt_reset(705): task not found

E (13843) task_wdt: esp_task_wdt_reset(705): task not found
E (13849) task_wdt: esp_task_wdt_reset(705): task not found
SPIFFS formatted successfully.
Openning GIF file from SD card...
Copy GIF in SPIFFS...
Starting animation...
GPIOViewer >> Release 1.5.9
GPIOViewer >> ESP32 Core Version 3.1.1
GPIOViewer >> Chip Model:ESP32-D0WD-V3, revision:301
GPIOViewer >> No PSRAM
GPIOViewer >> Web Application URL is: http://192.168.0.144:8080
GPIOViewer >> Connected, sampling interval is 100ms

Animation does not start. GPIO viewer is running, providing data, etc.

atlantikus, 2025-01-14:
When   //gpio_viewer.begin(); is commented as shown

```
[animated_gif_sdcard_spiffs.zip](https://github.com/user-attachments/files/18415620/animated_gif_sdcard_spiffs.zip)

```

Animation starts. Serial monitor shows following:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4688
load:0x40078000,len:15460
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3196
entry 0x400805a4
GPIOViewer >> Connecting to WiFi...
.GPIOViewer >> Connected to WiFi
SD card initialization...
Initialize SPIFFS...
Formatting SPIFFS...
E (2410) task_wdt: esp_task_wdt_reset(705): task not found
E (2411) task_wdt: esp_task_wdt_reset(705): task not found
E (2412) task_wdt: esp_task_wdt_reset(705): task not found
E (2441) task_wdt: esp_task_wdt_reset(705): task not found
E (2442) task_wdt: esp_task_wdt_reset(705): task not found
E (2442) task_wdt: esp_task_wdt_reset(705): task not found

~500 times E (xxx) task_wdt: esp_task_wdt_reset(705): task not found

E (13821) task_wdt: esp_task_wdt_reset(705): task not found
E (13826) task_wdt: esp_task_wdt_reset(705): task not found
E (13832) task_wdt: esp_task_wdt_reset(705): task not found
SPIFFS formatted successfully.
Openning GIF file from SD card...
Copy GIF in SPIFFS...
Starting animation...

thelastoutpostworkshop, 2025-01-14:
I will try your sketch on my side, I have the CYD board ESP32-2432S028

thelastoutpostworkshop, 2025-01-15:
your GIFDraw function is missing in your code, can you include it please, so i have the same code as you have to test

thelastoutpostworkshop, 2025-01-16:
here in this line 
`if (gif.open(filename, fileOpen, fileClose, fileRead, fileSeek, **GIFDraw**))`
GIFDraw is a function called inside gif.playFrame
You need to have one in your code or it won't compile.
Maybe zip you project and drop it in a comment below so I have everything to reproduce the problem.

atlantikus, 2025-01-16:
I already provided project in post 2 days ago [animated_gif_sdcard_spiffs.zip](https://github.com/user-attachments/files/18415620/animated_gif_sdcard_spiffs.zip). It compiles without problem, animation is running. If I uncomment line //gpio_viewer.begin(); animation does not start.

You can see here installed libraries
[image]

thelastoutpostworkshop, 2025-01-16:
In your screenshot, you have GIFDraw,ino, that's the file I need, it contains the GIFDraw function.

atlantikus, 2025-01-19:
Oh, yes, understood. Here it is 

[GIFDraw.zip](https://github.com/user-attachments/files/18469733/GIFDraw.zip)

thelastoutpostworkshop, 2025-01-19:
I was able to reproduce the problem I need to investigate further

thelastoutpostworkshop, 2025-01-19:
The bug is related to ADC pins reading which interfere with pins used for the display.
I will work on an update to the library soon.

The message `E (2442) task_wdt: esp_task_wdt_reset(705): task not found` that you see in the serial monitor is related to a [spiff issue](https://github.com/espressif/arduino-esp32/issues/10547)
I cannot do anything, it's an issue in the Arduino Core from Espressif.

thelastoutpostworkshop, 2025-01-26:
This is now fixed in v1.6.1

closed 2025-01-26 by thelastoutpostworkshop

atlantikus, 2025-01-28:
I tested with v1.6.1. My sketch now works with GPIOViewer. Thank you very much for fixing this bug.