thelastoutpostworkshop/gpio_viewer #154: Contributing boards

MatAtBread, 2025-01-17:
Not exactly an issue, but is the source of the static assets in Github, so people can contribute boards? My specific use case is the Seeed Xiao ESP32C6

Is it possible to parameterize the static HTML on the board so the ` <base href='https://thelastoutpostworkshop.github.io/microcontroller_devkit/gpio_viewer_1_5/'>` can point somewhere else for development?

MatAtBread, 2025-01-17:
I should have looked at the source code, of course the URL can be modified (duh!). https://github.com/thelastoutpostworkshop/gpio_viewer/blob/main/src/gpio_viewer.h#L28

thelastoutpostworkshop, 2025-01-17:
Of course you can contribute, the assets are here : https://github.com/thelastoutpostworkshop/microcontroller_devkit/tree/main/gpio_viewer_1_5

closed 2025-01-28 by thelastoutpostworkshop