thelastoutpostworkshop/gpio_viewer #157: failing to connect to wifi

simshouseengineering, 2025-01-19:
hey - I know this is likely an error on my end as no one else seems to be having issues. I followed your directions and had no issues... everything compiled and uploaded just fine. and yes I did alter the code to include my correct wifi credentials. I even double checked by commenting out your code and just including the wifi library and connecting myself, which compiled and connected just fine. So it's not my wifi or the controller.... it has to be the code, but it compiles fine, no errors. no leads. 

so when I run the program, I upload ONLY the example sketch as you provided.. and it seems to have trouble connecting to wifi and ends up in this loop inside the setup function..

thelastoutpostworkshop, 2025-01-19:
Show me the code that you include with the wifi library that is able to connect to your network

simshouseengineering, 2025-01-19:
Hang on.. I was following someone elses issues to see if it helped and I ended up messing up something else.. Unrelated issue.. Now my serial monitor is not working.. I'll reply with more information as soon as I can get this resolved.

thanks!

simshouseengineering, 2025-01-19:
Hey do you know what it means when the blink sketch wont compile? LEDBUILTin is unrecognized... I reinstalled the IDE and same issue.

thelastoutpostworkshop, 2025-01-19:
> Hey do you know what it means when the blink sketch wont compile? LEDBUILTin is unrecognized... I reinstalled the IDE and same issue.
> […](#)

it's not related to gpioviewer ?

simshouseengineering, 2025-01-19:
At this point I’m unsure.. nothing works now. I’m troubleshooting with
Arduino support.. was just wondering if you knew if that was a common error
or something.

thelastoutpostworkshop, 2025-01-19:
I was able to reproduce the problem, I need to investigate further.

simshouseengineering, 2025-01-19:
Hey I managed to get it working.. but I had to go as far as reinstalling windows haha. Anyway, now I’m good. Sent from my iPad

thelastoutpostworkshop, 2025-01-19:
ok thanks for letting me know

closed 2025-01-19 by thelastoutpostworkshop