thelastoutpostworkshop/gpio_viewer #161: Compiler Issue

F3rnw3h, 2025-02-22:
In file included from /private/var/folders/vf/fn3lh97d63x6xswvkmj325ym0000gn/T/.arduinoIDE-unsaved2025122-2783-va845e.o9z6b/gpioviewer/gpioviewer.ino:17:
/Users/.../Documents/Arduino/libraries/GPIOViewer/src/gpio_viewer.h: In member function 'void GPIOViewer::readADCPinsConfiguration()':
/Users/.../Documents/Arduino/libraries/GPIOViewer/src/gpio_viewer.h:590:33: error: no match for 'operator&' (operand types are 'volatile gpio_enable_reg_t' and 'int')
  590 |                 if (GPIO.enable & (1 << i))
      |                     ~~~~~~~~~~~ ^ ~~~~~~~~
      |                          |           |
      |                          |           int
      |                          volatile gpio_enable_reg_t

exit status 1

Compilation error: exit status 1

Board is DFRobot FireBeetle 2 ESP32-C6 and the code i used is the example code

thelastoutpostworkshop, 2025-02-23:
What is the ESP32 Arduino Espressif version you are using ?

F3rnw3h, 2025-02-23:
3.1.3

thelastoutpostworkshop, 2025-02-23:
It's now fixed in v1.6.2 which I pushed to the arduino library index, should be available in a couple of hours.

galimpic, 2025-02-24:
I tried it on S3 Zero and it works after the change, but on C3 Super Mini it only works with GPIO.enable.val, not GPIO.enable1.val

thelastoutpostworkshop, 2025-02-24:
It seams that the implementation of GPIO api is different for some boards, I will look into it

thelastoutpostworkshop, 2025-05-03:
This is now fixed in v1.6.3

closed 2025-05-03 by thelastoutpostworkshop