thelastoutpostworkshop/gpio_viewer #162: Is it possible to run this locally

Bendahon, 2025-03-10:
First up, love this project. I would like to add this to everything I do but it has some quirks, it would be nice to see it as an "all in one" package running offline. The 80k lines of CSS/JS scares me to look at but i'd love to see "Lite" version that lets me run locally... GPIO expanders like the MCP23017 would also be cool :)

thelastoutpostworkshop, 2025-03-10:
The web application is much larger than 80k with the images.  It could be run locally on a web server though.  Everything is open source so you can give it a try.

Bendahon, 2025-03-10:
I have some plans to try much simpler version I like it so much! Apologies if the wording was off as well, i meant the JS and CSS expanded to about 80,000 lines which seemed mad. Is there some library that it's using or is that just standard web dev today?

closed 2025-03-27 by thelastoutpostworkshop