thelastoutpostworkshop/gpio_viewer #166: Stop or End Gpio Viewer

dariods8474, 2025-03-27:
How to stop or end gpio viewer without powering off the device?

thelastoutpostworkshop, 2025-03-27:
There is currently no way to stop it since it's a tool to help troubleshoot problems.  Once troubleshoot is done, you comment out the library to deactivate it.

dariods8474, 2025-04-01:
I am creating a menu type option to be launched with a combination of buttons, where I would launch GPIO Viewer when needed and would like to stop it when not required. The library would always be present so could be used anytime for troubleshooting problems.
there is this line in your code that creates a task
`// Create a task for monitoring GPIOs
 xTaskCreate(&GPIOViewer::monitorTaskStatic, "GPIO Monitor Task", 4096, this, 1, NULL);`
Can this be used to stop the tool?
Gpio viewer does not show the status of ADC pins GPIO 32, 33, 34 and 35.

thelastoutpostworkshop, 2025-04-01:
Yes you can stop the task by deleting it like this:
```
TaskHandle_t gpioMonitorHandle = NULL;
xTaskCreate(&GPIOViewer::monitorTaskStatic, "GPIO Monitor Task", 4096, this, 1, &gpioMonitorHandle);
// ...
// When you want to stop the task:
vTaskDelete(gpioMonitorHandle);
gpioMonitorHandle = NULL; // optional: clear the handle
```

closed 2025-05-03 by thelastoutpostworkshop