thelastoutpostworkshop/gpio_viewer #169: Dependencies creating issues in PlatformIO

dariods8474, 2025-04-01:
The dependencies in the library.properties 
`depends=Async TCP,ESP Async WebServer`
are creating compilation issues in PlatformIO. Changing them to
`depends=AsyncTCP,ESPAsyncWebServer`
Removing the space between Asnc and TCP, space between ESP and Async and WebServer, solves the problem. It downloads the proper dependency libraries. no need to add in platformio.ini
`lib_deps =
    https://github.com/mathieucarbou/AsyncTCP.git
    https://github.com/mathieucarbou/AsyncTCP.git`

thelastoutpostworkshop, 2025-04-01:
Are you sure that's the right libraries downloaded when you remove the spaces ?
If you look at the the name given by the author of these libraries in the file library.properties they contain spaces :
https://github.com/mathieucarbou/AsyncTCP
https://github.com/mathieucarbou/ESPAsyncWebServer

dariods8474, 2025-04-01:
Using Async TCP and ESP Async WebServer were giving errors. Deleting them would result in re downloading them at compiling. If internet was disconnected to prevent download, the download would fail with http error and compiling would stop. So I downloaded AsyncTCP(without space) and EspAsyncWebServer (without spaces), edited the GPIO viewer library.properties file and deleted Async TCP and ESP Async WebServer folder and it compiled without errors. I will check if only editing the library.properties is helping in downloading the right libraries. I will revert back

thelastoutpostworkshop, 2025-04-01:
I am not sure what you are trying to achieve, the GPIOViewer library is tested against the libraries specified in its library.properties, if you are using others it may not work.

closed 2025-04-12 by thelastoutpostworkshop