thelastoutpostworkshop/gpio_viewer #170: Please add ESP32 C3 dev kit M-1

DavidP1958, 2025-04-01:
Hi,

Can you please add ESP32 C3 dev kit M-1

[https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp32-c3-devkitm-1/user_guide.html](url)

Thanks a lot.

Dave

thelastoutpostworkshop, 2025-04-01:
Your link does not work.
Please provide a link to a picture of the board and a link to the pinout.

DavidP1958, 2025-04-02:
Hi,

Sorry about the link, don't know what happened

I have been back to the Espressif web site and have several options, I hope
one works....

   1. Online version
   <https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp32-c3-devkitm-1/index.html>
   2. Link shown when I opened the document
   <https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp-dev-kits-en-master-esp32c3.pdf>
   3. A copy I saved to my OneDrive account
   <https://1drv.ms/b/c/f39d0e59ea943689/EReyYbpwkAJGqkcaaocfRxoB21FmLULS5FkwZ9-dQT5V7A?e=nKHW9y>

The board is the ESP32-C3-DevKitM-1 and is in section 2 of the document.

Please let me know if things work this time or if there are still problems,
I can also send it as an attachment, but I think that is too big for
GitHib  to handle.

Regards

Dave

thelastoutpostworkshop, 2025-04-12:
The board is now available as ESP32-C3 Mini-1

closed 2025-04-12 by thelastoutpostworkshop