thelastoutpostworkshop/gpio_viewer #180: Fails to connect to Wi-Fi network; console error messages

raneil86, 2025-07-18:
I am just beginning to learn to use the ESP32, and your project will be invaluable to me as I learn.  However, I cannot seem to get it working and I'm not sure if I am missing something or perhaps it's a compatibility issue with the specific ESP32 I'm using.  

I am using the Freenove ESP32 WROVER (specifically, I bought this beginner kit from Amazon: https://a.co/d/1R2ROLv).  I am using the simple Wi-Fi Client tutorial sketch included with the kit.  It works fine, and connects to my Wi-Fi network and obtains an IP address from the router via DHCP, as expected.

However, when I add the gpio_viwer code as demonstrated in your YouTube video, the console displays the following in a repeating loop:

21:43:07.055 -> GPIOViewer >> Release 1.6.3
21:43:07.055 -> GPIOViewer >> ESP32 Core Version 3.2.1
21:43:07.055 -> GPIOViewer >> Chip Model:ESP32-D0WD-V3, revision:301
21:43:07.087 -> GPIOViewer >> PSRAM Size 4.00 MB
21:43:07.087 -> 
21:43:07.087 -> assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1854 (Required to lock TCPIP core functionality!)
21:43:07.087 -> 
21:43:07.087 -> 
21:43:07.087 -> Backtrace: 0x40082668:0x3ffb1f50 0x4008d515:0x3ffb1f70 0x40093779:0x3ffb1f90 0x400f55a3:0x3ffb20d0 0x400f5719:0x3ffb20f0 0x400dd670:0x3ffb2110 0x400dc801:0x3ffb2160 0x400d386b:0x3ffb2180 0x400d3e1a:0x3ffb2210 0x400e1e77:0x3ffb2270 0x4008e206:0x3ffb2290
21:43:07.124 -> 
21:43:07.124 -> 
21:43:07.124 -> 
21:43:07.124 -> 
21:43:07.124 -> ELF file SHA256: 95dcdf70d
21:43:07.124 -> 
21:43:07.192 -> Rebooting...
21:43:07.192 -> ets Jul 29 2019 12:21:46
21:43:07.192 -> 
21:43:07.192 -> rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
21:43:07.225 -> configsip: 0, SPIWP:0xee
21:43:07.225 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:43:07.225 -> mode:DIO, clock div:1
21:43:07.225 -> load:0x3fff0030,len:4888
21:43:07.225 -> load:0x40078000,len:16456
21:43:07.225 -> load:0x40080400,len:4
21:43:07.225 -> load:0x40080404,len:3476
21:43:07.225 -> entry 0x400805b4

For reference, here is the entire sketch:

#include <gpio_viewer.h> // Must me the first include in your project
#include <WiFi.h>

GPIOViewer gpio_viewer;

const char *ssid_Router     = "Sidewinder"; //Enter the router name
const char *password_Router = "creak97_puck"; //Enter the router password

void setup(){
  Serial.begin(115200);
  delay(2000);
  Serial.println("Setup start");
  WiFi.begin(ssid_Router, password_Router);
  Serial.println(String("Connecting to ")+ssid_Router);
  while (WiFi.status() != WL_CONNECTED){
    delay(500);
    Serial.print(".");
  }
  Serial.println("\nConnected, IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("Setup End");
  gpio_viewer.begin();
}
 
void loop() {

}

raneil86, 2025-07-18:
Problem resolved: apparently had multiple copies of the Async TCP Library installed.  The gpio_viewer is now working properly.  Love it!

thelastoutpostworkshop, 2025-07-18:
Great!

closed 2025-07-18 by thelastoutpostworkshop