thelastoutpostworkshop/gpio_viewer #186: Please Include i2C PCF8575 expansion modules

Russ5612, 2025-09-26:
The GPIO Viewer is an excellent enhancement idea, thank you very much.

[image]

Please add the I2C PCF8575 I/O Expansion Modules, I have numerous projects that use multiple IO's and use up to 6 modules connected to a single ESP32

Many thanks
Russ

thelastoutpostworkshop, 2025-09-26:
Thank you, so Pxx refers to GPIO xx (ex.: P12 is GPIO 12), is that right ?

Russ5612, 2025-10-04:
You are correct.
These boards expand the GPIO of an ESP32 using the I2C bus.
See this PDF for more details https://www.ti.com/lit/ds/symlink/pcf8575.pdf 

[pcf8575 (1).pdf](https://github.com/user-attachments/files/22702817/pcf8575.1.pdf)

So to summarize, the GPIO for each PCF8575 board is the same, the ESP32 works on the various board ID's to identify the specific GPIO

Some of my code:
1. Each board is given an id, below is YAML code which I use in ESPHome:
	pcf8574:
	  - id: 'pcf8575_0' # Board 0
		address: 0x20
		i2c_id: bus_b
		pcf8575: True
	  - id: 'pcf8575_1' # Board 1
		address: 0x21
		i2c_id: bus_b
		pcf8575: True
	  - id: 'pcf8575_2' # Board 2
		address: 0x22
		i2c_id: bus_b
		pcf8575: True    
	  - id: 'pcf8575_3' # Board 3
		address: 0x23
		i2c_id: bus_b
		pcf8575: True
I use them as extra inputs or outputs. Below is an abbreviated YAML code which I use in ESPHome, note "#" is for comments as "//" in Arduino IDE:		
binary_sensor:
  - platform: gpio
    #restore_mode: RESTORE_INVERTED_DEFAULT_OFF
    name: "PCF8575 Pin #20-00 - Motion Sensor A"
    pin:
      pcf8574: pcf8575_0 # Board 0
      # Use pin number 00
      number: 00
      # One of INPUT or OUTPUT
      mode: INPUT
      inverted: true

  - platform: gpio
    name: "PCF8575 Pin #20-01 - Motion Sensor B"
    pin:
      pcf8574: pcf8575_0 # Board 0
      # Use pin number 01
      number: 01
      # One of INPUT or OUTPUT
      mode: INPUT
      inverted: true

- platform: gpio
    name: "PCF8575 Pin #21-00 - Switch Study 1=on/0=off"
    pin:
      pcf8574: pcf8575_1 # Board 1
      # Use pin number 00
      number: 00
      # One of INPUT or OUTPUT
      mode: INPUT
      inverted: true
  
  - platform: gpio
    name: "PCF8575 Pin #21-01 - Sonoff Lounge 1=on/0=off"
    pin:
      pcf8574: pcf8575_1 # Board 1
      # Use pin number 01
      number: 01
      # One of INPUT or OUTPUT
      mode: INPUT
      inverted: true
	  
  - platform: gpio
    name: "PCF8575 Pin #22-00 - Lamp D  1=on/0=off"
    pin:
      pcf8574: pcf8575_2 # Board 2
      # Use pin number 00
      number: 00
      # One of INPUT or OUTPUT
      mode: INPUT
      inverted: true
  
  - platform: gpio
    name: "PCF8575 Pin #22-01 - Lamp E  1=on/0=off"
    pin:
      pcf8574: pcf8575_2 # Board 2
      # Use pin number 01
      number: 01
      # One of INPUT or OUTPUT
      mode: INPUT
      inverted: true	  
	  
switch:

  - platform: gpio
    name: "PCF8575 Pin #20-08 - Entrance Light"
    pin:
      pcf8574: pcf8575_0 # Board 0
      # Use pin number 08
      number: 08
      # One of INPUT or OUTPUT
      mode: OUTPUT
      inverted: true
	  
  - platform: gpio      
    name: "PCF8575 Pin #22-08 - Power to Alarm"
    pin:
      pcf8574: pcf8575_3 # Board 3
      # Use pin number 08
      number: 08
      # One of INPUT or OUTPUT
      mode: OUTPUT
      inverted: true

Russ5612, 2025-10-04:
Sorry the positing did some strange font adjustments